jeanlemotan/esp32-cam-fpv #3: What is the limiting factor in the current system?

jtomes123, 2021-07-15:
Hi, what is currently the limiting factor of the whole system? Is it the camera or the esp32 itself, would offloading image processing and acquisition to another board allow lower latency at higher resolutions?

diegovazquez, 2021-08-21:
For what i know, the limiting factor is the hardware. Sensor and bandwidth.
 
More info: https://github.com/espressif/esp32-camera/issues/15

jeanlemotan, 2022-02-14:
Indeed, the limiting factors are the sensor (it limits the max FPS achievable and the quality is not that great), sensor->esp32 i2s bandwidth (limits the max FPS and compression, limits the number of GPIO left for other things) and esp32 wifi bandwidth (this is a tradeoff between range, bandwidth and redundancy: you can usually choose one or two of these 3 variables, but you do have full freedom to choose what you want here).

closed 2022-02-14 by jeanlemotan