jeanlemotan/esp32-cam-fpv #4: fatal error: crc.h: No such file or directory

jeng37, 2022-01-13:
root@raspberrypi:~# git clone https://github.com/jeanlemotan/esp32-cam-fpv.git
Cloning into 'esp32-cam-fpv'...
remote: Enumerating objects: 277, done.
remote: Counting objects: 100% (277/277), done.
remote: Compressing objects: 100% (185/185), done.
remote: Total 277 (delta 99), reused 238 (delta 74), pack-reused 0
Receiving objects: 100% (277/277), 905.21 KiB | 388.00 KiB/s, done.
Resolving deltas: 100% (99/99), done.
root@raspberrypi:~# cd esp32-cam-fpv
root@raspberrypi:~/esp32-cam-fpv# cd gs/
root@raspberrypi:~/esp32-cam-fpv/gs# ls
Makefile  src
root@raspberrypi:~/esp32-cam-fpv/gs# make
g++ -MT .o/src/main.o -MD -MP -MF .d/src/main.Td -std=c++17 -O3 -DNDEBUG -ffast-math -funroll-loops -mcpu=cortex-a8 -mfpu=neon -Wall -DRASPBERRY_PI -Isrc -Isrc/utils -Isrc/imgui -Icomponents/common -I/opt/vc/include/ -I/usr/include/freetype2 -c -o .o/src/main.o src/main.cpp
src/main.cpp:20:10: fatal error: crc.h: No such file or directory
   20 | #include "crc.h"
      |          ^~~~~~~
compilation terminated.
make: *** [Makefile:126: .o/src/main.o] Error 1
root@raspberrypi:~/esp32-cam-fpv/gs#

peteallenm, 2022-01-23:
Looking at the Makefile, I would guess you have to change every line that has components/ to ../components/
Untested

closed 2022-02-14 by jeanlemotan

careyer, 2022-03-17:
This issue still persists... can somebody please advice how to get rid of the 

```
src/main.cpp:20:10: fatal error: crc.h: No such file or directory
   20 | #include "crc.h"
      |          ^~~~~~~

```
compile error?

JulesThuillier, 2023-01-09:
@careyer : the answer is in the messages above.