jeanlemotan/esp32-cam-fpv #5: Allwinner v3s + ESP32 version

peteallenm, 2022-01-23:
I love the premise of this project, but want to take it a step up in video quality and latency.
To do this I'm using a Allwinner V3s as camera interface, image signal processor and jpeg compressor. This allows many many more sensors, frame rates and resolutions to be an option, then using an ESP32 as wifi broadcast card (as per ESP32_bridge_broadcast).

Currently testing with an OV5640 as it works, but I'm aiming to use  a  OV4669 (same as the Runcam Split 3), which should provide a big boost in quality, and be good for 720p/60 (the sensor is good for 720p/120, but I don't have enough CSI lanes)
I've eventually managed to get hardware accelerated jpeg encoding to run, and it should be good for 720p 60fps.

Wifi wise I'm ending up with similar findings to what you found with the ESP8266_bridge_broadcast : finding SDIO Wifi modules capable of Monitor mode is tough, getting drivers on the ancient kernel I'm using a pain, so I heading down using an ESP32 over SPI (tested to 25Mhz, and 4K SPI messages allows over 20Mbps of throughput).
I just need to hack esp32-cam-fpv to take it's input from SPI, and then get the OV4669 working and it should be good for a test.

Have you got any interest on working on esp32-cam-fpv, or are you done with it? If possible I'd love a bit of assistance in working out the magic in camera_data_available(), but if not I'm sure I'll get there eventually.

jeanlemotan, 2022-02-14:
Hey, 
I'm glad you find this useful, but unfortunately I had to pause the project due to some other projects taking too much of my time.
I'd be glad to help out with that function - essentially it's taking data from the camera - chunk by chunk - and adding it to the outgoing packet. Then, once the packet is done, it's sent to the FEC encoder which will output several more packets for data redundancy and that's pretty much it.

If you're interested in using the esp32 just for sending the data, check out my other project that does just this: https://github.com/jeanlemotan/esp32_bridge_broadcast
It comes with a C++ lib as well that takes care of talking to the esp32 from a raspberry pi or similar.

Let me know if you need anything. I will probably have some latency in answering though. Good luck!

closed 2022-02-14 by jeanlemotan

Ncerzzk, 2022-02-15:
hey, I'm also interested with the idea, and my idea is almost the same with yours.
while I have some advice, may be you can try using the esp32 as the sdio slave, and there is a official repo of esp32:https://github.com/espressif/esp-hosted
of courese the repo seems to not support the monitor mode, so there should be some work to do.