jeanlemotan/esp32-cam-fpv #7: Is it possible to use this code for two way audio-video platform ?

happytm, 2022-02-15:
Is it possible to use this code to make a video walkie talkie using wifi-broadcast like what's app  ?

Thanks.

jeanlemotan, 2022-05-11:
Sure, it's very possible. Hook up a screen and you'll be able to display the incoming video stream.
Potential issues:
1. GPIO. You'll need another I2S peripheral to hook up the screen and I'm not sure you have enough pins left.
2. JPEG decompression: You'll have to decompress the incoming JPEGs and this takes time, and more importantly - memory. A solution would be to reduce resolution to 320x240 for example and use PSRAM for this but I doubt you'll get more than 10 FPS

closed 2022-05-13 by jeanlemotan

happytm, 2022-05-13:
Thank you for your comment.

happytm, 2022-05-13:
10 FPS for long range video walkie talkie for areas with spotty cellular coverage is a good compromise.

Is it possible to use Smartphones on both sides to use it's screens,microphones and speaker to view video like it is used in following project (audio only walkie talkie using lora and codec2)?:

https://github.com/sh123/codec2_talkie

Thanks.