jeanlemotan/esp32-cam-fpv #15: Is there any long voyage test results?

lida2003, 2023-01-20:
AI Thinker camera uses PCB antenna or ipx， is there any test on long voyage?

jeanlemotan, 2023-01-21:
Only brief tests where I walk around the my apartment building with the GS, while the TX is on the 5th floor in a pretty busy wifi area. I got around 300m line of sight before the packet loss was big enough to get choppy video. But the noise floor was super high because of all the wifi networks around.

closed 2023-01-21 by jeanlemotan

lida2003, 2023-01-22:
Maybe PA + RF antenna should be used. Thanks for the info.

JulesThuillier, 2023-02-14:
Same here, I got around 250m with LOS.

lida2003, 2023-02-14:
> I got around 250m with LOS.

PCB antenna? or ipx external antenna?

JulesThuillier, 2023-02-14:
External antenna, I think it was a 3dBi antenna.
I am going to run more tests with different antennas and an amplifier soon. I am trying to achieve 1km :o

lida2003, 2023-02-15:
[image]
@JulesThuillier  This kind of 3dbi ipx antenna right? BTW how many wifi cards do you use dring this 250m range (LOS) test?