jeanlemotan/esp32-cam-fpv #16: esp-idf 4.3 + latest code(56ee43e)  constantly reboot

lida2003, 2023-01-20:
See below code for ESP_IDF and esp32-cam-fpv.

I use AI-Thinker module with 2MB PSRAM(which should be causion). The code constantly rebooted, any clues?

```
$ git log -n 1
commit 56ee43ec5ea813809fc1a9da1cc644559b4ea975 (HEAD -> release/v4.3, origin/release/v4.3)
Merge: 0d48889e81 b44da528db
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Tue Jan 17 19:56:03 2023 +0800

    Merge branch 'bugfix/sta_add_config_for_wpa3_transition_disable_v4.3' into 'release/v4.3'

    esp_wifi:Add wifi station config for enabling transition_disbale feature

    See merge request espressif/esp-idf!21332

$ git log -n 1
commit 9610792f8544361bac21f774c7700551f9bcfc6e (HEAD -> main, origin/main, origin/HEAD)
Author: jeanlemotan <10252034+jeanlemotan@users.noreply.github.com>
Date:   Thu Apr 29 20:40:40 2021 +0200

    Update README.md

```

Just flash and monitor the logs:

```
$ idf.py -p /dev/ttyUSB0 flash monitor
Executing action: flash
Running ninja in directory /home/daniel/Work/esp32-cam-fpv/air_firmware/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/daniel/Work/esp-sdk/esp-idf-v4.3/components/esptool_py && /usr/bin/.../home/daniel/Work/esp-sdk/esp-idf-v4.3/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 air_firmware.bin
esptool.py v3.3.2-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:5a:1b:6b:d5:74
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x000adfff...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 231.8 kbit/s)...
Hash of data verified.
Compressed 26192 bytes to 16369...
Writing at 0x00001000... (100 %)
Wrote 26192 bytes (16369 compressed) at 0x00001000 in 1.2 seconds (effective 172.4 kbit/s)...
Hash of data verified.
Compressed 646352 bytes to 422177...
Writing at 0x00010000... (3 %)
Writing at 0x0001ac67... (7 %)
Writing at 0x00026bac... (11 %)
Writing at 0x0002c454... (15 %)
Writing at 0x000319f2... (19 %)
Writing at 0x00037198... (23 %)
Writing at 0x0003c3fa... (26 %)
Writing at 0x0004198c... (30 %)
[... 680 lines cut ...]
  At 0x3ffb83c8 len 162872 free 36852 allocated 123528 min_free 36772
    largest_free_block 35840 alloc_blocks 85 free_blocks 2 total_blocks 87
  At 0x3ffe0440 len 15072 free 12920 allocated 0 min_free 12920
    largest_free_block 12800 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 111688 allocated 0 min_free 111688
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 2189903 allocated 193196 min_free 2189823 largest_free_block 1998848
Initialized
I (1117) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1127) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1136) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1145) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1155) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1164) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1173) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1183) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1192) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1202) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1211) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1221) sccb: pin_sda 26 pin_scl 27

I (1225) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

```

jeanlemotan, 2023-01-21:
I didn't test with the latest version, but the stack overflow indicates that the stack size has to be increased for the Wifi TX task. Probably some internals have changed in the latest IDF and more stack is being used. Try to add 512 bytes.

There is another error in the log: wifi:invalid interface 0
Not sure what could be causing this though.

lida2003, 2023-01-22:
Hummm... I think` wifi:invalid interface 0` might be also comes from esp-idf.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6532
load:0x40078000,len:15292
load:0x40080400,len:4276
0x40080400: _init at ??:?

entry 0x4008067c
I (27) boot: ESP-IDF v4.3.4-332-g56ee43ec5e 2nd stage bootloader
I (27) boot: compile time 12:59:25
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
E (44) qio_mode: Failed to set QIE bit, not enabling QIO mode
I (50) boot.esp32: SPI Speed      : 80MHz
I (55) boot.esp32: SPI Mode       : DIO
I (59) boot.esp32: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) boot_comm: chip revision: 3, min. application chip revision: 0
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=114d4h ( 70868) map
I (144) esp_image: segment 1: paddr=000214fc vaddr=3ffb0000 size=03b10h ( 15120) load
I (149) esp_image: segment 2: paddr=00025014 vaddr=40080000 size=0b004h ( 45060) load
I (165) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6c844h (444484) map
I (298) esp_image: segment 4: paddr=0009c86c vaddr=4008b004 size=11434h ( 70708) load
I (337) boot: Loaded app from partition at offset 0x10000
I (337) boot: Disabling RNG early entropy source...
I (348) psram: This chip is ESP32-D0WD
I (349) spiram: Found 16MBit SPI RAM device
I (349) spiram: SPI RAM mode: flash 80m sram 80m
I (354) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (361) cpu_start: Pro cpu up.
I (365) cpu_start: Starting app cpu, entry point is 0x4008134c
0x4008134c: call_start_cpu1 at /home/daniel/Work/esp-sdk/esp-idf-v4.3/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (623) spiram: SPI SRAM memory test OK
I (630) cpu_start: Pro cpu start user code
I (630) cpu_start: cpu freq: 240000000
I (630) cpu_start: Application information:
I (633) cpu_start: Project name:     air_firmware
I (639) cpu_start: App version:      47ba7e7-dirty
I (644) cpu_start: Compile time:     Jan 21 2023 13:16:47
I (650) cpu_start: ELF file SHA256:  849688d28797ba90...
I (656) cpu_start: ESP-IDF:          v4.3.4-332-g56ee43ec5e
I (663) heap_init: Initializing. RAM available for dynamic allocation:
I (670) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (676) heap_init: At 3FFB83C8 len 00027C38 (159 KiB): DRAM
I (682) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (688) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (695) heap_init: At 4009C438 len 00003BC8 (14 KiB): IRAM
I (701) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (709) spi_flash: detected chip: gd
I (713) spi_flash: flash io: qio
I (718) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Initializing...
MEMORY at start:
Heap summary for capabilities 0x00000004:
  At 0x3f800000 len 2097151 free 2094999 allocated 0 min_free 2094999
[... 87 lines cut ...]
  At 0x3ffb83c8 len 162872 free 348 allocated 159900 min_free 332
    largest_free_block 264 alloc_blocks 118 free_blocks 3 total_blocks 121
  At 0x3ffe0440 len 15072 free 7792 allocated 5120 min_free 7792
E (1616) wifi:    largest_free_block 7680 alloc_blocks 2 free_blocks 1 total_blocks 3
  At 0x3ffe4350 len 113840 free 103492 allocated 8192 min_free 103492
    largest_free_block 102400 alloc_blocks 1 free_blocks 1 total_blocks 2
invalid interface 0
  Totals:
E (1657) wifi:invalid interface 0
    free 2140075 allocated 242880 min_free 2140059 largest_free_block 1998848
WLAN S: E (0, R: 016, E6: 4,1)  D: 0, wi% :f 9i9 :|| FPS:in v3, alD: i22824d i5 || D: nt0, E: erfa0
ce 0
E (1677) wifi:invalid interface 0
E (1680) wifi:invalid interface 0
E (1683) wifi:invalid interface 0
E (1686) wifi:invalid interface 0
E (1689) wifi:invalid interface 0
E (1692) wifi:invalid interface 0
E (1695) wifi:invalid interface 0
E (1698) wifi:invalid interface 0
E (1701) wifi:invalid interface 0
E (1704) wifi:invalid interface 0
E (1707) wifi:invalid interface 0

```

lida2003, 2023-01-22:
It seems ok {WiFi mode(STA=1)， interface(STA=0)， package buffer length(1500)}

- ESP-IDF v4.3.4-332-g56ee43ec5e
- latest esp32-cam-fpv
- 

```
I (27) boot: ESP-IDF v4.3.4-332-g56ee43ec5e 2nd stage bootloader
I (27) boot: compile time 16:22:01
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
E (44) qio_mode: Failed to set QIE bit, not enabling QIO mode
I (50) boot.esp32: SPI Speed      : 80MHz
I (55) boot.esp32: SPI Mode       : DIO
I (59) boot.esp32: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) boot_comm: chip revision: 3, min. application chip revision: 0
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=113b0h ( 70576) map
I (144) esp_image: segment 1: paddr=000213d8 vaddr=3ffb0000 size=03b44h ( 15172) load
I (149) esp_image: segment 2: paddr=00024f24 vaddr=40080000 size=0b0f4h ( 45300) load
I (165) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6bab4h (441012) map
I (297) esp_image: segment 4: paddr=0009badc vaddr=4008b0f4 size=1141ch ( 70684) load
I (336) boot: Loaded app from partition at offset 0x10000
I (336) boot: Disabling RNG early entropy source...
I (347) psram: This chip is ESP32-D0WD
I (348) spiram: Found 16MBit SPI RAM device
I (348) spiram: SPI RAM mode: flash 80m sram 80m
I (353) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (360) cpu_start: Pro cpu up.
I (364) cpu_start: Starting app cpu, entry point is 0x4008135c
0x4008135c: call_start_cpu1 at /home/daniel/Work/esp-sdk/esp-idf-v4.3/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (622) spiram: SPI SRAM memory test OK
I (629) cpu_start: Pro cpu start user code
I (629) cpu_start: cpu freq: 240000000
I (629) cpu_start: Application information:
I (632) cpu_start: Project name:     air_firmware
I (638) cpu_start: App version:      47ba7e7-dirty
I (643) cpu_start: Compile time:     Jan 22 2023 16:21:47
I (649) cpu_start: ELF file SHA256:  2133162de0175013...
I (655) cpu_start: ESP-IDF:          v4.3.4-332-g56ee43ec5e
I (662) heap_init: Initializing. RAM available for dynamic allocation:
I (669) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (675) heap_init: At 3FFB83D0 len 00027C30 (159 KiB): DRAM
I (681) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (687) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (694) heap_init: At 4009C510 len 00003AF0 (14 KiB): IRAM
I (700) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (708) spi_flash: detected chip: gd
I (712) spi_flash: flash io: qio
I (717) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Initializing...
MEMORY at start:
Heap summary for capabilities 0x00000004:
  At 0x3f800000 len 2097151 free 2094999 allocated 0 min_free 2094999
    largest_free_block 2064384 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 4 allocated 4132 min_free 4
    largest_free_block 0 alloc_blocks 36 free_blocks 0 total_blocks 36
  At 0x3ffb83d0 len 162864 free 131032 allocated 29560 min_free 130096
    largest_free_block 129024 alloc_blocks 30 free_blocks 1 total_blocks 31
  At 0x3ffe0440 len 15072 free 12920 allocated 0 min_free 12920
[... 91 lines cut ...]
esp_wifi_80211_tx if 0 size 1500
  Totals:
    free 2140067 allocated 242880 min_free 2140051 largest_free_block 1998848
iLAN SE: 0 , R: (10,6 E: 79)3, D: 0 wi, % : 99 f|| i:iFPSn: 3, D: 2871va64 || D: 0, E: l0
d interface 0
esp_wifi_80211_tx if 0 size 1500
E (1699) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1705) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1711) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1717) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1723) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1729) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1735) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1741) wifi:invalid interface 0
esp_wifi_80211_tx if 0 size 1500
E (1747) wifi:invalid interface 0

```

lida2003, 2023-01-23:
`Initialized WiFi mode(1), if(0)` /`esp_wifi_80211_tx if 0 size 1500` shows wifi mode STA(1), IF STA(0) , packet size(1500)
Currently, it responds ESP_ERR_WIFI_IF, it might be `wifi_interface_t ifx` error.
Since esp_wifi_80211_tx is a library (maybe from libnet80211.a), I didn't know much about internal mechanism.

```
/**
  * @brief     Send raw ieee80211 data
  *
  * @attention Currently only support for sending beacon/probe request/probe response/action and non-QoS
  *            data frame
  *
  * @param     ifx interface if the Wi-Fi mode is Station, the ifx should be WIFI_IF_STA. If the Wi-Fi
  *            mode is SoftAP, the ifx should be WIFI_IF_AP. If the Wi-Fi mode is Station+SoftAP, the
  *            ifx should be WIFI_IF_STA or WIFI_IF_AP. If the ifx is wrong, the API returns ESP_ERR_WIFI_IF.
  * @param     buffer raw ieee80211 buffer
  * @param     len the length of raw buffer, the len must be <= 1500 Bytes and >= 24 Bytes
  * @param     en_sys_seq indicate whether use the internal sequence number. If en_sys_seq is false, the
  *            sequence in raw buffer is unchanged, otherwise it will be overwritten by WiFi driver with
  *            the system sequence number.
  *            Generally, if esp_wifi_80211_tx is called before the Wi-Fi connection has been set up, both
  *            en_sys_seq==true and en_sys_seq==false are fine. However, if the API is called after the Wi-Fi
  *            connection has been set up, en_sys_seq must be true, otherwise ESP_ERR_WIFI_ARG is returned.
  *
  * @return
  *    - ESP_OK: success
  *    - ESP_ERR_WIFI_IF: Invalid interface
  *    - ESP_ERR_INVALID_ARG: Invalid parameter
  *    - ESP_ERR_WIFI_NO_MEM: out of memory
  */

esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);

```

JulesThuillier, 2023-01-23:
Are you using the same esp-idf version as described in the README ? I had issue with that and now I am using release/v4.3 branch (commit 389a848a6a7b11cfd2e33bf34752483906bc9237 from December 29 09:42) of ESP IDF and it works.

I also had to change line 693 of main.cpp for the following line :
`esp_err_t err = esp_wifi_config_80211_tx_rate(ESP_WIFI_IF, (wifi_phy_rate_t)rates[(int)value]);`

rottaran, 2023-01-23:
Do you get the 802.11n modes (MCS0 to MCS7) working? For me, `esp_wifi_config_80211_tx_rate` with these rates never sent anything in my test programs on IDF 4.4 and 5.0. Today I got it working by setting the bandwidth, the channel, and then the rate:

```
ESP_ERROR_CHECK(esp_wifi_set_bandwidth(ESP_WIFI_IF, WIFI_BW_HT20));
ESP_ERROR_CHECK(esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE));
ESP_ERROR_CHECK(esp_wifi_config_80211_tx_rate(ESP_WIFI_IF, WIFI_PHY_RATE_MCS1_LGI));
```

Also got 40MHz rates working:

```
ESP_ERROR_CHECK(esp_wifi_set_bandwidth(ESP_WIFI_IF, WIFI_BW_HT40));
ESP_ERROR_CHECK(esp_wifi_set_channel(9, WIFI_SECOND_CHAN_ABOVE));
ESP_ERROR_CHECK(esp_wifi_config_80211_tx_rate(ESP_WIFI_IF, WIFI_PHY_RATE_MCS1_LGI));
```

Without setting the bandwidth, it did not work. Maybe because the default is HT40 or maybe it is initializing something internally.

lida2003, 2023-01-24:
@JulesThuillier @rottaran 
IDF 4.3 4.4 latest code is OK for sending packet out, with `esp_err_t err = esp_wifi_config_80211_tx_rate(ESP_WIFI_IF, (wifi_phy_rate_t)rates[(int)value]);` properly set.

BUT no video on RPI 3B+ nor Ubuntu laptop. please see  latest code(9610792） no video on RPI Window #22 

@jeanlemotan Do you have an suggestion on this?

lida2003, 2023-01-25:
@rottaran I have got IDF 4.4 (latest) working (I mean compile and log shows working)

@JulesThuillier I just changed code your suggested and some debug issues.

@jeanlemotan This is my PR:  Fix no WiFi RF packet send issue and upgrade to IDF v4.4(latest) #23   . If you can see the video on gs. Then I think it's OK to merge.

BTW: Please let usknow:

1) If this PR works
2) Please try latest gs code
3) are you using Pi 4 hardware?

PS: I wanna know why my gs is not working :(

jeanlemotan, 2023-01-26:
I'll try this weekend to check the stuff above.
I am using a Pi 4 indeed, but I have a Pi 3 as well and I'll try with it to see what happens.

lida2003, 2023-01-26:
> I'll try this weekend to check the stuff above.
> I am using a Pi 4 indeed, but I have a Pi 3 as well and I'll try with it to see what happens.

Thanks. that's really helps.

JulesThuillier, 2023-01-28:
@lida2003 Sorry no I don't have patches for 5.0, currently I'm still trying to figure out how to make it work completely (I still just have a black screen pop up with a crash soon after on my RPi 4), before upgrading the soft.

lida2003, 2023-01-29:
> I still just have a black screen pop up with a crash soon after on my RPi 4

@JulesThuillier  Hummm....  RPI 4,  now we are on the same starting line.

It seems that RPI 3B+/4/Ubuntu 20.04 have the black screen with latest gs code. Maybe we have to dig a little bit deeper.

Good news is  @jeanlemotan will take a look when he has time.

jeanlemotan, 2023-02-05:
The air firmware should now work with IDF4.4.4. But pls use the committed sdkconfig, it's critical for memory reasons.
I just tested now and video is send at high FPS.

closed 2023-02-05 by jeanlemotan

lida2003, 2023-02-06:
fresh build

$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/daniel/Work/esp32-cam-fpv-main/air_firmware/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/daniel/Work/esp32-cam-fpv-main/air_firmware"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
Updating lock file at /home/daniel/Work/esp32-cam-fpv-main/air_firmware/dependencies.lock
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/freemodbus.
-- Project sdkconfig file /home/daniel/Work/esp32-cam-fpv-main/air_firmware/sdkconfig
-- Found Python3: /home/daniel/.espressif/python_env/idf4.4_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "air_firmware" version: df74c7b
-- Adding linker script /home/daniel/Work/esp32-cam-fpv-main/air_firmware/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/soc/esp32/ld/esp32.peripherals.ld
-- DEBUG: Use esp-modbus component folder: /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/freemodbus.
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap common console cxx driver efuse esp-tls esp32 esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/app_trace /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/app_update /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/asio /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/bootloader /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/bootloader_support /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/bt /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/cbor /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/cmock /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/coap /home/daniel/Work/esp32-cam-fpv-main/components/common /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/console /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/cxx /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/driver /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/efuse /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp-tls /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp32 /home/daniel/Work/esp32-cam-fpv-main/air_firmware/components/esp32-camera /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_adc_cal /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esp_common /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/ [... cut ...]
I (464) cpu_start: ESP-IDF:          v4.4.4
I (469) heap_init: Initializing. RAM available for dynamic allocation:
I (476) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

Done
daniel@daniel-ThinkPad-SL410:~/Work/esp32-cam-fpv-main/air_firmware$ git log -n 1
commit df74c7b83cd673cb5a32e1275c29e025a35b55b4 (HEAD -> jeanlemotan-main, origin/jeanlemotan-main, origin/HEAD)
Author: jeanlemotan <10252034+jeanlemotan@users.noreply.github.com>
Date:   Sun Feb 5 22:38:21 2023 +0100

    Changes sdkconfig for IDF4.4.4