jeanlemotan/esp32-cam-fpv #20: gs compiles ok, but it doesn't work properly

lida2003, 2023-01-22:
I have to wifi cards installed (which has monitor mode)

```
pi@PiNas:~ $ iw phy1 info
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        RTS threshold: 2347
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x186e
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 150 Mbps
                HT TX/RX MCS rate indexes supported: 0-7, 32
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps
                        * 5.5 Mbps
                        * 11.0 Mbps
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
[... 237 lines cut ...]

And gs has been compiled OK, but it failed during execution, any ideas?

```
esp32-cam-fpv/gs $ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 125: Initializing pigpio
(I) src/PI_HAL.cpp: 283: Drivers: 4
(I) src/PI_HAL.cpp: 286: Driver 0: x11
(I) src/PI_HAL.cpp: 286: Driver 1: wayland
(I) src/PI_HAL.cpp: 286: Driver 2: RPI
(I) src/PI_HAL.cpp: 286: Driver 3: dummy
(I) src/PI_HAL.cpp: 295: Mode 0: 1280x720
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile vertex shader!
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile fragment shader!
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to link shader program!
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile vertex shader!
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile fragment shader!
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to link shader program!
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlan1 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Comms.cpp: 581: Opening interface wlan2 in monitor mode
(E) src/Comms.cpp: 623: Error in pcap_activate: That device is not up

```

lida2003, 2023-01-23:
It seems USB power issue with two cards, reducing to one.

Still 100% CPU and no popup window.

[image]

lida2003, 2023-01-23:
Still error in ImGui_ImplOpenGL3_NewFrame, popup window (NOT Full size).

[image]

lida2003, 2023-01-23:
glGetShaderiv issue, but I don't know how to fix.

[image]

lida2003, 2023-01-23:
OK, I now know the issue. Cause you are using Pi4(OpenGL 3.0), I'm using Pi3b+ (OpenGL 2.0). 

[RPI OpenGL versions](https://forums.raspberrypi.com/viewtopic.php?t=294077)

I'll provide a patch, when I get " esp-idf 4.3 + latest code(56ee43e) constantly reboot #16 " works.

JulesThuillier, 2023-01-23:
Hello !
I'm running exactly into the same problem.
Also if I compile using SDL flag, it runs, receives packets but does not display anything on my screen (I'm using Bullseye with desktop). And if I disable SDL if get a `terminate called after throwing an instance of 'fmt::v7::format_error'` just after network initialisation...
At least it's good to see a fresh post on that project, I already feel a bit less lost !

lida2003, 2023-01-24:
@JulesThuillier Haha, Thansk.
I got esp-idf 4.3 (389a848a6a7b11cfd2e33bf34752483906bc9237 ) , and compiled the way your provided. Although, I have got the same issue "black screen, send and receive ok"

The gs program seems something wrong with OpenGL 2.0.

```
$ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 125: Initializing pigpio
(I) src/PI_HAL.cpp: 283: Drivers: 4
(I) src/PI_HAL.cpp: 286: Driver 0: x11
(I) src/PI_HAL.cpp: 286: Driver 1: wayland
(I) src/PI_HAL.cpp: 286: Driver 2: RPI
(I) src/PI_HAL.cpp: 286: Driver 3: dummy
(I) src/PI_HAL.cpp: 295: Mode 0: 1280x720
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlan1 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Comms.cpp: 581: Opening interface wlan2 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Video_Decoder.cpp: 160: SDL window: 7244056
(I) src/Video_Decoder.cpp: 160: SDL window: 7244056
(I) src/Video_Decoder.cpp: 160: SDL window: 7244056
(I) src/Video_Decoder.cpp: 160: SDL window: 7244056
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 252840, RSSI: -71, Latency: 60/60/60
(I) src/Video_Decoder.cpp: 333: Texture: 2
(I) src/Video_Decoder.cpp: 333: Texture: 3
(I) src/Video_Decoder.cpp: 333: Texture: 4
(E) src/Video_Decoder.cpp: 349: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, output.pbo)file src/Video_Decoder.cpp line 349
(E) src/Video_Decoder.cpp: 362: GL error 1280 in glBufferData(GL_PIXEL_UNPACK_BUFFER, pbo_size, nullptr, GL_STREAM_DRAW)file src/Video_Decod                            er.cpp line 362
(E) src/Video_Decoder.cpp: 386: GL error 1280 in glBindTexture(GL_TEXTURE_2D, output.textures[i])file src/Video_Decoder.cpp line 386
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 395: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, 0)file src/Video_Decoder.cpp line 395
(I) src/Video_Decoder.cpp: 333: Texture: 5
(I) src/Video_Decoder.cpp: 333: Texture: 6
(I) src/Video_Decoder.cpp: 333: Texture: 7
(E) src/Video_Decoder.cpp: 349: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, output.pbo)file src/Video_Decoder.cpp line 349
(E) src/Video_Decoder.cpp: 362: GL error 1280 in glBufferData(GL_PIXEL_UNPACK_BUFFER, pbo_size, nullptr, GL_STREAM_DRAW)file src/Video_Decod                            er.cpp line 362
(E) src/Video_Decoder.cpp: 386: GL error 1280 in glBindTexture(GL_TEXTURE_2D, output.textures[i])file src/Video_Decoder.cpp line 386
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 395: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, 0)file src/Video_Decoder.cpp line 395
(I) src/Video_Decoder.cpp: 333: Texture: 8
(I) src/Video_Decoder.cpp: 333: Texture: 9
(I) src/Video_Decoder.cpp: 333: Texture: 10
(E) src/Video_Decoder.cpp: 349: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, output.pbo)file src/Video_Decoder.cpp line 349
(E) src/Video_Decoder.cpp: 362: GL error 1280 in glBufferData(GL_PIXEL_UNPACK_BUFFER, pbo_size, nullptr, GL_STREAM_DRAW)file src/Video_Decod                            er.cpp line 362
(E) src/Video_Decoder.cpp: 386: GL error 1280 in glBindTexture(GL_TEXTURE_2D, output.textures[i])file src/Video_Decoder.cpp line 386
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
[... cut ...]
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 395: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, 0)file src/Video_Decoder.cpp line 395
^C2023-01-24 09:23:46 sigHandler: Unhandled signal 2, terminating

terminate called without an active exception
2023-01-24 09:23:46 sigHandler: Unhandled signal 6, terminating

2023-01-24 09:23:46 gpioWrite: pigpio uninitialised, call gpioInitialise()
double free or corruption (fasttop)
2023-01-24 09:23:46 sigHandler: Unhandled signal 6, terminating

```

jeanlemotan, 2023-01-26:
The GL errors are super relevant since it's used for rendering, so getting any error in there means you'll most probably get no rendering at all.

jeanlemotan, 2023-01-26:
Why not use GL3.0 on Ubuntu?

lida2003, 2023-01-26:
> Why not use GL3.0 on Ubuntu?

Error logs were from RPI 3B+. Ubuntu is my laptop, there is no error, but black screen also.