jeanlemotan/esp32-cam-fpv #22: latest code(9610792） no video on RPI Window

lida2003, 2023-01-23:
latest code(9610792）+ ESP_IDF 4.3 beta 1 + AI Thinker Module (2MB PSRAM)==> build OK
latest code(9610792）+ RPI 3B+  minor OPENGL 2.0 and on wlan1 for TxRx ==> build OK

no video on RPI Window, all black.

[image]

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:6220
load:0x40078000,len:15076
load:0x40080400,len:4316
0x40080400: _init at ??:?

entry 0x40080674
I (29) boot: ESP-IDF v4.3-beta1 2nd stage bootloader
I (29) boot: compile time 14:47:29
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) qio_mode: Enabling default flash chip QIO
E (45) qio_mode: Failed to set QIE bit, not enabling QIO mode
I (51) boot.esp32: SPI Speed      : 80MHz
I (56) boot.esp32: SPI Mode       : DIO
I (60) boot.esp32: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (108) boot_comm: chip revision: 3, min. application chip revision: 0
I (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=14094h ( 82068) map
I (149) esp_image: segment 1: paddr=000240bc vaddr=3ffb0000 size=03b70h ( 15216) load
I (155) esp_image: segment 2: paddr=00027c34 vaddr=40080000 size=00404h (  1028) load
I (155) esp_image: segment 3: paddr=00028040 vaddr=40080404 size=07fd8h ( 32728) load
I (175) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=73454h (472148) map
I (322) esp_image: segment 5: paddr=000a347c vaddr=400883dc size=146ech ( 83692) load
I (368) boot: Loaded app from partition at offset 0x10000
I (368) boot: Disabling RNG early entropy source...
I (379) psram: This chip is ESP32-D0WD
I (380) spiram: Found 16MBit SPI RAM device
I (380) spiram: SPI RAM mode: flash 80m sram 80m
I (385) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (392) cpu_start: Pro cpu up.
I (396) cpu_start: Starting app cpu, entry point is 0x40081340
0x40081340: call_start_cpu1 at /home/daniel/Work/esp-sdk/esp-idf-v4.3-beta1/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (653) spiram: SPI SRAM memory test OK
I (661) cpu_start: Pro cpu start user code
I (661) cpu_start: cpu freq: 240000000
I (661) cpu_start: Application information:
I (664) cpu_start: Project name:     air_firmware
I (669) cpu_start: App version:      47ba7e7-dirty
I (675) cpu_start: Compile time:     Jan 23 2023 14:47:18
I (681) cpu_start: ELF file SHA256:  017c16ca42493aa1...
I (687) cpu_start: ESP-IDF:          v4.3-beta1
I (692) heap_init: Initializing. RAM available for dynamic allocation:
I (699) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (705) heap_init: At 3FFB8000 len 00028000 (160 KiB): DRAM
I (712) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (718) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (724) heap_init: At 4009CAC8 len 00003538 (13 KiB): IRAM
I (731) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (739) spi_flash: detected chip: gd
[... 143 lines cut ...]
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
```

lida2003, 2023-01-23:
two wifi cards, still nothing.

```
$ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 125: Initializing pigpio
(I) src/PI_HAL.cpp: 283: Drivers: 4
(I) src/PI_HAL.cpp: 286: Driver 0: x11
(I) src/PI_HAL.cpp: 286: Driver 1: wayland
(I) src/PI_HAL.cpp: 286: Driver 2: RPI
(I) src/PI_HAL.cpp: 286: Driver 3: dummy
(I) src/PI_HAL.cpp: 295: Mode 0: 1280x720
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlan1 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Comms.cpp: 581: Opening interface wlan2 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Video_Decoder.cpp: 160: SDL window: 28891200
(I) src/Video_Decoder.cpp: 160: SDL window: 28891200
(I) src/Video_Decoder.cpp: 160: SDL window: 28891200
(I) src/Video_Decoder.cpp: 160: SDL window: 28891200
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
```

lida2003, 2023-01-24:
Really strange thing.
1) No video string
2) Camera received gs, but gs can receive streaming video intermittently
3) Just run for a little bit long time, gs received no packet any more......

[image]
[image]

lida2003, 2023-01-24:
Much more stable with 4.3 (56ee43ec5ea813809fc1a9da1cc644559b4ea975)

```
$ git log -n 1
commit 56ee43ec5ea813809fc1a9da1cc644559b4ea975 (HEAD, origin/release/v4.3)
Merge: 0d48889e81 b44da528db
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Tue Jan 17 19:56:03 2023 +0800

    Merge branch 'bugfix/sta_add_config_for_wpa3_transition_disable_v4.3' into 'release/v4.3'

    esp_wifi:Add wifi station config for enabling transition_disbale feature

    See merge request espressif/esp-idf!21332

```

[image]

lida2003, 2023-01-24:
OK, might be RF signal issue.

Just focused on black screen issue. Any idea?

```
$ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 125: Initializing pigpio
(I) src/PI_HAL.cpp: 283: Drivers: 4
(I) src/PI_HAL.cpp: 286: Driver 0: x11
(I) src/PI_HAL.cpp: 286: Driver 1: wayland
(I) src/PI_HAL.cpp: 286: Driver 2: RPI
(I) src/PI_HAL.cpp: 286: Driver 3: dummy
(I) src/PI_HAL.cpp: 295: Mode 0: 1280x720
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlan1 in monitor mode
(I) src/Comms.cpp: 282: DLT_IEEE802_11_RADIO Encap
(I) src/Video_Decoder.cpp: 160: SDL window: 16305032
(I) src/Video_Decoder.cpp: 160: SDL window: 16305032
(I) src/Video_Decoder.cpp: 160: SDL window: 16305032
(I) src/Video_Decoder.cpp: 160: SDL window: 16305032
(I) src/main.cpp: 104: RX len: 1470, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 7350, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 2940, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 2940, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 2940, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 1470, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 5880, RSSI: -49, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 0, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 5880, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 7350, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 2940, RSSI: -49, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 1470, RSSI: 0, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 8820, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 1470, RSSI: -50, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 214620, RSSI: -46, Latency: 87/87/87
(I) src/Video_Decoder.cpp: 333: Texture: 2
(I) src/Video_Decoder.cpp: 333: Texture: 3
(I) src/Video_Decoder.cpp: 333: Texture: 4
(E) src/Video_Decoder.cpp: 349: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, output.pbo)file src/Video_Decoder.cpp line 349
(E) src/Video_Decoder.cpp: 362: GL error 1280 in glBufferData(GL_PIXEL_UNPACK_BUFFER, pbo_size, nullptr, GL_STREAM_DRAW)file src/Video_Decod                            er.cpp line 362
(E) src/Video_Decoder.cpp: 386: GL error 1280 in glBindTexture(GL_TEXTURE_2D, output.textures[i])file src/Video_Decoder.cpp line 386
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 389: GL error 1281 in glTexImage2D(GL_TEXTURE_2D, 0, GL_R8, width, height, 0, GL_RED, GL_UNSIGNED_BYTE, (void*)of                            fset)file src/Video_Decoder.cpp line 389
(E) src/Video_Decoder.cpp: 395: GL error 1280 in glBindBuffer(GL_PIXEL_UNPACK_BUFFER, 0)file src/Video_Decoder.cpp line 395
(I) src/main.cpp: 104: RX len: 637980, RSSI: -47, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 627690, RSSI: -46, Latency: 0/0/0
(I) src/main.cpp: 104: RX len: 186690, RSSI: -49, Latency: 15/15/15
(I) src/main.cpp: 104: RX len: 8820, RSSI: -50, Latency: 49/49/49
(I) src/main.cpp: 104: RX len: 4410, RSSI: -49, Latency: 2/124/63
(I) src/main.cpp: 104: RX len: 2940, RSSI: -50, Latency: 2/2/2
(I) src/main.cpp: 104: RX len: 4410, RSSI: 0, Latency: 148/183/166
(I) src/main.cpp: 104: RX len: 10290, RSSI: -50, Latency: 10/38/24
(I) src/main.cpp: 104: RX len: 1470, RSSI: -50, Latency: 170/170/170
(I) src/main.cpp: 104: RX len: 5880, RSSI: -50, Latency: 60/188/124
```

lida2003, 2023-01-24:
Does it have something to do with OpenGL ES version?  

[image]

lida2003, 2023-01-24:
Tried my laptop(ubuntu) , still no video.

[image]

whstudio123, 2023-01-31:
I struggled with these codes all day and I have some new findings: When gl_flush () is turned on, it can "work normally". Then I read the code of [the official demo of imgui](https://github.com/ocornut/imgui/blob/master/examples/example_sdl_opengl3/main.cpp) and our code line by line. Are there any bugs nearby? (Because I noticed there are some location differences with the official demo)
[image]
[image]
[image]

BTW:In order to facilitate debugging and modification, I changed it to window mode.And the program can only run at very low resolution and speed.

lida2003, 2023-02-01:
The good news is that there is an image with glFlush(), although not 30 FPS at 800x600.

:)

[30 FPS at 800x600](https://user-images.githubusercontent.com/10252034/116135308-43c08c00-a6d1-11eb-99b9-9dbb106b6489.mp4)

whstudio123, 2023-02-01:
Not bad.So....Do you have a try ? is there a wrong url you upload? I think you want to upload your screenshoot instead.hiahia
[image]

whstudio123, 2023-02-01:
And I found that the card type is **soooo important**. Because i setup two different cards in two computers and tcpdump them .One is RTL8812AU (TP-LINK T4U).The other is EDUP EP-N8508GS.
**Suprisingly I found the EDUP EP-N8508GS captured well but T4U not**.(Howerer T4U and RTL8812au cards are well known by OpenHD or ZE-BroardCast players.) EDUP EP-N8508GS only has low speed sniff and injection.maybe So thats why i can only achieved very very low frame rate.
Which card do you use ?Could you share some infos?

lida2003, 2023-02-01:
> Not bad.So....Do you have a try ? is there a wrong url you upload? I think you want to upload your screenshoot instead.hiahia

Well, That's @jeanlemotan 's test video. I have the same video quality as you do.

rottaran, 2023-02-01:
> And I found that the card type is **soooo important**. Because i setup two different cards in two computers and tcpdump them .One is RTL8812AU (TP-LINK T4U).

You need to install a special driver for these RTL chipsets. For example https://github.com/morrownr/8812au or https://github.com/aircrack-ng/rtl8812au The in-kernel driver barely works.

whstudio123, 2023-02-01:
We need to back to the gl_flush() code and the window fresh process. Why could this magic happen?(I am a beginner to opengl and imgui.)

lida2003, 2023-02-01:
> You need to install a special driver for these RTL chipsets

@rottaran @jeanlemotan 

How about following wifi card？It has monitor mode, but I'm NOT sure if it'll casue performance issue?

08:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 03)

JulesThuillier, 2023-02-01:
Hello !
I am having the same dirty video as you.... And the controls are unusable, can you guys move the sliders ?

lida2003, 2023-02-01:
> And the controls are unusable, can you guys move the sliders ?

NO

whstudio123, 2023-02-01:
> move

No, I can't move the slides either.

whstudio123, 2023-02-01:
New progress. Now i get about 10fps with low but affordable (for me) video.
The key is channel and data rate.  

Tcpdump do shows the real data rate. That means when it shows 1.0Mbps. To get affordable video, you must increase data rate.（ I increased them to 24Mbps） Must double check them.

The maximum data rate do has a relationship with card type and driver. Make an experiment for example, my EDUP card can only allow 1.0Mbps monitor. So when i send 24Mbps packet , The card still shows other packets in 1.0Mbps and capture nothing of my 24Mbps packet.

The rate adjudgment API in air_firmware may have bugs.  I change IDF version , write something hardcoded and successfully adjust rate（ can be really seen and dump in 24Mbps）.

Things get better step by step. I will post some procedures after I entirely figure out the api .:)

whstudio123, 2023-02-01:
New progress. Now i get about 10fps with low but affordable (for me) video.
The key is channel and data rate.  

Tcpdump do shows the real data rate. That means when it shows 1.0Mbps. To get affordable video, you must increase data rate.（ I increased them to 24Mbps） Must double check them.

The maximum data rate do has a relationship with card type and driver. Make an experiment for example, my EDUP card can only allow 1.0Mbps monitor. So when i send 24Mbps packet , The card still shows other packets in 1.0Mbps and capture nothing of my 24Mbps packet.

Data rate is NOT higher is better because it maybe over the card maximum monitor rate.I make another experiment to explain this. My t4u card could receive 54Mbps esp32 packets but it drops some packets.When it comes to 24Mbps ,it works well.And i learned from a paper that monitor speed is almost all quite low . Card above 20Mbps is a quite good card for monitor.

The rate adjudgment API in air_firmware may have bugs.  I change IDF version , write something hardcoded and successfully adjust rate（ can be really seen and dump in 24Mbps）.

Things get better step by step. I will post some procedures after I entirely figure out the api .:)

JulesThuillier, 2023-02-02:
I am going to work on those issues today, can you share some of your progress ? (especially where you changed the data rate in the air firmware).

whstudio123, 2023-02-02:
> I am going to work on those issues today, can you share some of your progress ? (especially where you changed the data rate in the air firmware).

I will make an detail experiment about the api in two hours, double check and post the result.

whstudio123, 2023-02-02:
The experiment comes.**I use IDF-4.4.3.**

use internal API:
[image]

The result:
[image]

use esp_wifi_config_80211_tx_rate() API:
[image]

The result:
[image]
(Pay attention to the raw data , we could also see the JFIF header which means jpeg.haha~)
**ATTENTION! Version matters!**
I checked ESP32 API doc version by version and find that **we could officially adjust the rate of 802.11 injection only when using IDF-4.4.2 or newer.**
IDF-4.4.2 and above says:
https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/wifi.html#wi-fi-80211-packet-send
[image]

IDF-4.4.1 and below says:
https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-guides/wifi.html#wi-fi-80211-packet-send
[image]

I also find something in ESP-FAQ about the internal api:
https://docs.espressif.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/wifi.html
[image]

So, Maybe ESP-FAQ is a little bit of confusing."To set and fix the Wi-Fi sending rate" may only just about "wifi"(we normally heared from) but not 802.11 raw packet.
**No matter what the ESP-FAQ says,just following the API doc and using the official API to adjust 802.11 rate without a problem.**
I will make a PR and note we should use at least IDF-4.4.2 version.

lida2003, 2023-02-02:
> at least IDF-4.4.2 version

Hummm.... @jeanlemotan 's video use esp-idf-v4.3-beta1. It seems IDF version depends. 

Anyway, it has been made a great progress  :)

whstudio123, 2023-02-02:
> > at least IDF-4.4.2 version
> 
> Hummm.... @jeanlemotan 's video use esp-idf-v4.3-beta1. It seems IDF version depends.
> 
> Anyway, it has been made a great progress :)

Yes.In todays' v4.3.1 doc ,rate adjustment is **officially** not support.
I also do the experiment in IDF-4.3.2 which is close enough to jeanlemotan's version.
I have two esp32-cam . One flash internal API+ IDF-4.3.2 ,the other flash esp_wifi_config_80211_tx_rate() + IDF-4.4.3.
**The result is the internal api cannot adjust the rate in IDF-4.3.2. It even loss a lot of packets in tcpdump.  And the chip get crazy hot (sucks more than 500mA)**
However,when using esp_wifi_config_80211_tx_rate() + IDF-4.4.3, only sucks about 250mA and packet works well.
So ,as it is not work at all for me . Just forget about the internal api , move to higher idf ,following the API doc and using the official API to adjust 802.11 rate without a problem.

whstudio123, 2023-02-02:
Install a lot of IDF this week &#x1F603;
[image]

rottaran, 2023-02-02:
whstudio123, can you try `esp_wifi_config_80211_tx_rate(ESP_WIFI_IF, true, WIFI_PHY_RATE_MCS0_LGI)`? This is index 14 in the `rates` array. My suspicion is, that it does not, unless you set the channel bandwidth correctly:

```
ESP_ERROR_CHECK(esp_wifi_set_bandwidth(ESP_WIFI_IF, WIFI_BW_HT20));
ESP_ERROR_CHECK(esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE));
```

My memory might be wrong. I thought that the bitrate was adjusted automatically. But I could not find the code in main.cpp.

whstudio123, 2023-02-02:
All right,i will try it now.
I write some ideas on your "esp32_bridge_broadcast" repo.Could you plz have a look?

whstudio123, 2023-02-02:
Just change 10 to 14 and other code are stayed the same.
[image]

rottaran, 2023-02-02:
interesting. It seems to work without changes then. Still don't understand why it did not work in my own software. maybe there is a hidden firmware difference between ESP32 and ESP32-S3 too

whstudio123, 2023-02-02:
Maybe.I am using ESP32.

whstudio123, 2023-02-03:
> > move
> 
> No, I can't move the slides either.

New progress.Only one line change in PI_HAL.cpp could let us use mouse:
[image]

jeanlemotan, 2023-02-05:
Finally I managed to get some free time and compile the GS & Air firmware and test.
I used initially the IDF 5.0, but I got some compile errors in the camera component, and after reading the migration document I decided to go back to 4.X. I think it will take some serious work to port the camera component to 5.0 since they changed the low level drivers and ATM I don't see any advantage.
So back to IDF 4.4.4
The air firmware worked but only sent a few seconds of data, after which it stopped sending. It till received fine, but no video was sent. Then I remember that I spend a significant amount of time to tweak the sdkconfig to free as much memory as possible and speed up the drivers by placing callbacks in IRAM etc, so after 2h of tweaking and testing, I got the air config to send continuously. The new sdkconfig is committed now.

As for the GS, I didn't get anything rendered except for the imgui window borders. I tested in X11 and there is clearly an issue with the rendering, but I'll have to debug tomorrow more.

It should be trivial to fix though.

whstudio123, 2023-02-06:
> Finally I managed to get some free time and compile the GS & Air firmware and test. I used initially the IDF 5.0, but I got some compile errors in the camera component, and after reading the [...]

Yes.DO NOT DEL the sdkconfg !!!! There are serval settings especially about IRAM and wifi stacks.

lida2003, 2023-02-06:
> Yes.DO NOT DEL the sdkconfg !!!! There are serval settings especially about IRAM and wifi stacks.

Actually, I didn't find anything special between sdkconfig(auto generated) and git version.

If this project is truely sensitvie to some default settings, please try "sdkconfig.defaults"