jeanlemotan/esp32-cam-fpv #24: Undefined reference issues when compiling gs code.

whstudio123, 2023-01-29:
- I could not compile the gs code.I have tried my best to compile them and tried to modify the Makefile but nothing worked.there is  some screenshots when i tried to compile :

```
g++ -o gs "-L/usr/lib/aarch64-linux-gnu" "-L/usr/lib" "-L/opt/vc/lib" -lstdc++ -lm -pthread -lz -lrt -lfreetype -lSDL2 -lturbojpeg -lpcap -lGLESv2 -lpigpio  .o/src/main.o .o/src/droid_sans_font.o .o/src/HUD.o .o/src/imgui_impl_opengl3.o .o/src/PI_HAL.o .o/src/Comms.o .o/src/Video_Decoder.o .o/src/utils/radiotap/radiotap.o .o/src/imgui/imgui_impl_sdl.o .o/src/imgui/imgui_demo.o .o/src/imgui/imgui_draw.o .o/src/imgui/imgui.o .o/src/imgui/misc/freetype/imgui_freetype.o .o/../components/common/crc.o .o/../components/common/fec.o .o/src/fmt/format.o .o/src/fmt/os.o
/usr/bin/ld: .o/src/main.o: in function `comms_thread_proc()':
main.cpp:(.text+0xb14): undefined reference to `gpioWrite'
/usr/bin/ld: main.cpp:(.text+0xd10): undefined reference to `gpioWrite'
/usr/bin/ld: .o/src/main.o: in function `main':
main.cpp:(.text.startup+0x94): undefined reference to `gpioSetMode'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::prepare_filter(Comms::PCap&)':
Comms.cpp:(.text+0x34e0): undefined reference to `pcap_datalink'
/usr/bin/ld: Comms.cpp:(.text+0x3620): undefined reference to `pcap_compile'
/usr/bin/ld: Comms.cpp:(.text+0x3634): undefined reference to `pcap_setfilter'
/usr/bin/ld: Comms.cpp:(.text+0x3644): undefined reference to `pcap_freecode'
/usr/bin/ld: Comms.cpp:(.text+0x364c): undefined reference to `pcap_get_selectable_fd'
/usr/bin/ld: Comms.cpp:(.text+0x36f4): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x375c): undefined reference to `pcap_freecode'
/usr/bin/ld: Comms.cpp:(.text+0x3768): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::prepare_pcap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Comms::PCap&)':
Comms.cpp:(.text+0x3b38): undefined reference to `pcap_create'
/usr/bin/ld: Comms.cpp:(.text+0x3b48): undefined reference to `pcap_set_snaplen'
/usr/bin/ld: Comms.cpp:(.text+0x3b58): undefined reference to `pcap_set_promisc'
/usr/bin/ld: Comms.cpp:(.text+0x3b68): undefined reference to `pcap_set_rfmon'
/usr/bin/ld: Comms.cpp:(.text+0x3b78): undefined reference to `pcap_set_timeout'
/usr/bin/ld: Comms.cpp:(.text+0x3b88): undefined reference to `pcap_set_immediate_mode'
/usr/bin/ld: Comms.cpp:(.text+0x3b9c): undefined reference to `pcap_set_buffer_size'
/usr/bin/ld: Comms.cpp:(.text+0x3bdc): undefined reference to `pcap_activate'
/usr/bin/ld: Comms.cpp:(.text+0x3c00): undefined reference to `pcap_setdirection'
/usr/bin/ld: Comms.cpp:(.text+0x3c2c): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x3c70): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x3cac): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x3d24): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x3d78): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Comms.o:Comms.cpp:(.text+0x3dc8): more undefined references to `pcap_geterr' follow
```
(all the libraries are undefined reference.)

- The instructions on compile SDL2 are outdated.When doing the clone , i got SDL3 and the newer version needs cmake ,not configure.
- 64bit OS support is needed.maybe means a faster compute speed?
- We do need to move to a newer version of IDF (5.0) . Because the support of 80211 raw radio doesn't include dumping the CRC error frame in IDF 4.3. We still need the error frames because we have the ability to correct them by fec. 
[wi-fi-80211-packet-send](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-guides/wifi.html#wi-fi-80211-packet-send)
  ```
  The following packets will NOT be dumped to the application:
      802.11 error frame, such as the frame with a CRC error, etc.
  ```
  I notice that IDF 5.0 allows us to do this. 

If all the above things really act like what i say,i am very very glad to help to fix them if anyone helps me to compile the gs code :)?

jeanlemotan, 2023-01-30:
What platform are you compiling on? It looks like it's missing libpcap and pigpio.

whstudio123, 2023-01-31:
I only paste a few lines of the log.Actually all the libraries are undefined reference （even the pthread）.The platform is rpi4 with the lastest 64 bit OS raspbian.I do know i need 32 bit OS ,however i tried rpi3 with 32 and 64 OS,rpi4 with 32 and 64 OS ,but all of them stucked in undefied reference.

- step0:install OS to SD card
- step1:sudo apt-get update
- step2:sudo apt-get upgrade
- step3:sudo apt install libdrm-dev libgbm-dev libgles2-mesa-dev libpcap-dev libturbojpeg0-dev libts-dev libsdl2-dev
- step4:git clone esp32-cam-fpv
- step5:cd ./gs
- step6:make -j4

whstudio123, 2023-01-31:
- Step0 : Install OS to SD card : I use RPI Imager v1.6.2 to auto download and flash the **32 Bit Bullseye RaspberryPi OS** to SD card.

[image]

- Step1 : sudo apt-get update : 
[image]

- Step2 : sudo apt-get upgrade : 
[image]

- Step3 : sudo reboot
- Step4 : sudo apt install libdrm-dev libgbm-dev libgles2-mesa-dev libpcap-dev libturbojpeg0-dev libts-dev libsdl2-dev : 
[image]

- Step5 : git clone https://github.com/jeanlemotan/esp32-cam-fpv.git : 
[image]

- Step6 : cd ./esp32-cam-fpv/gs and make -j4
```
mlinux@raspberrypi:~/esp32-cam-fpv/gs $ make -j4
g++ -L/usr/lib -L=/opt/vc/lib -lstdc++ -lm -lpthread -lz -lrt -lfreetype -lSDL2 -lturbojpeg -lpcap -lGLESv2 -lpigpio  -o gs .o/src/main.o .o/src/droid_sans_font.o .o/src/HUD.o .o/src/imgui_impl_opengl3.o .o/src/PI_HAL.o .o/src/Comms.o .o/src/Video_Decoder.o .o/src/utils/radiotap/radiotap.o .o/src/imgui/imgui_impl_sdl.o .o/src/imgui/imgui_demo.o .o/src/imgui/imgui_draw.o .o/src/imgui/imgui.o .o/src/imgui/misc/freetype/imgui_freetype.o .o/../components/common/crc.o .o/../components/common/fec.o .o/src/fmt/format.o .o/src/fmt/os.o
/usr/bin/ld: .o/src/main.o: in function `run()':
main.cpp:(.text+0x294): undefined reference to `pthread_create'
/usr/bin/ld: main.cpp:(.text+0x29c): undefined reference to `pthread_create'
/usr/bin/ld: .o/src/main.o: in function `comms_thread_proc()':
main.cpp:(.text+0xef4): undefined reference to `gpioWrite'
/usr/bin/ld: main.cpp:(.text+0xf18): undefined reference to `gpioWrite'
/usr/bin/ld: .o/src/main.o: in function `main':
main.cpp:(.text.startup+0x80): undefined reference to `gpioSetMode'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::prepare_filter(Comms::PCap&)':
Comms.cpp:(.text+0x55b0): undefined reference to `pcap_datalink'
/usr/bin/ld: Comms.cpp:(.text+0x5734): undefined reference to `pcap_compile'
/usr/bin/ld: Comms.cpp:(.text+0x5748): undefined reference to `pcap_setfilter'
/usr/bin/ld: Comms.cpp:(.text+0x5758): undefined reference to `pcap_freecode'
/usr/bin/ld: Comms.cpp:(.text+0x5760): undefined reference to `pcap_get_selectable_fd'
/usr/bin/ld: Comms.cpp:(.text+0x581c): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x58ac): undefined reference to `pcap_freecode'
/usr/bin/ld: Comms.cpp:(.text+0x58b4): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::prepare_pcap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Comms::PCap&)':
Comms.cpp:(.text+0x5d04): undefined reference to `pcap_create'
/usr/bin/ld: Comms.cpp:(.text+0x5d18): undefined reference to `pcap_set_snaplen'
/usr/bin/ld: Comms.cpp:(.text+0x5d2c): undefined reference to `pcap_set_promisc'
/usr/bin/ld: Comms.cpp:(.text+0x5d40): undefined reference to `pcap_set_rfmon'
/usr/bin/ld: Comms.cpp:(.text+0x5d54): undefined reference to `pcap_set_timeout'
/usr/bin/ld: Comms.cpp:(.text+0x5d68): undefined reference to `pcap_set_immediate_mode'
/usr/bin/ld: Comms.cpp:(.text+0x5d80): undefined reference to `pcap_set_buffer_size'
/usr/bin/ld: Comms.cpp:(.text+0x5dc0): undefined reference to `pcap_activate'
/usr/bin/ld: Comms.cpp:(.text+0x5de4): undefined reference to `pcap_setdirection'
/usr/bin/ld: Comms.cpp:(.text+0x5e04): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x5e38): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x5e50): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x5ee0): undefined reference to `pcap_geterr'
/usr/bin/ld: Comms.cpp:(.text+0x5f60): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Comms.o:Comms.cpp:(.text+0x5fd0): more undefined references to `pcap_geterr' follow
/usr/bin/ld: .o/src/Comms.o: in function `Comms::tx_thread_proc()':
Comms.cpp:(.text+0xe5d4): undefined reference to `pcap_inject'
/usr/bin/ld: Comms.cpp:(.text+0xe71c): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::enumerate_interfaces[abi:cxx11]()':
Comms.cpp:(.text+0x12128): undefined reference to `pcap_findalldevs'
/usr/bin/ld: Comms.cpp:(.text+0x122a0): undefined reference to `pcap_freealldevs'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::init(Comms::RX_Descriptor const&, Comms::TX_Descriptor const&)':
Comms.cpp:(.text+0x18660): undefined reference to `pthread_create'
/usr/bin/ld: Comms.cpp:(.text+0x18668): undefined reference to `pthread_create'
/usr/bin/ld: Comms.cpp:(.text+0x186b0): undefined reference to `pthread_create'
/usr/bin/ld: Comms.cpp:(.text+0x186bc): undefined reference to `pthread_create'
/usr/bin/ld: .o/src/Comms.o: in function `Comms::process_rx_packet(Comms::PCap&)':
Comms.cpp:(.text+0x1a92c): undefined reference to `pcap_next_ex'
/usr/bin/ld: Comms.cpp:(.text+0x1b530): undefined reference to `pcap_geterr'
/usr/bin/ld: .o/src/Video_Decoder.o: in function `Video_Decoder::~Video_Decoder()':
Video_Decoder.cpp:(.text+0x72c8): undefined reference to `glDeleteTextures'
[... 365 lines cut ...]
/usr/bin/ld: imgui_freetype.cpp:(.text+0x1504): undefined reference to `FT_Done_Face'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x1510): undefined reference to `FT_Done_FreeType'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x152c): undefined reference to `FT_Done_Face'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x1538): undefined reference to `FT_Done_FreeType'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x1550): undefined reference to `FT_Done_Face'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x155c): undefined reference to `FT_Done_FreeType'
[... cut ...]
/usr/bin/ld: imgui_freetype.cpp:(.text+0x160c): undefined reference to `FT_GlyphSlot_Oblique'
/usr/bin/ld: imgui_freetype.cpp:(.text+0x1618): undefined reference to `FT_GlyphSlot_Embolden'
collect2: error: ld returned 1 exit status
make: *** [Makefile:115: gs] Error 1
```
[image]

And here is my uname info:
[image]

I can find my libraries here:
[image]

I also add this path to Makefile but nothing changes:
[image]

**I can makesure all the downloads&installs&updates&upgrades are successful. So.....Any idea:)?**

lida2003, 2023-01-31:
Latest code of gs, RPI 3B+ with following image and apt-get upgrade, should be fine :)

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
$ uname -r
5.10.103-v7+
$ uname -a
Linux PiNas 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
```

whstudio123, 2023-01-31:
That means i need  RPI 3B+ and 32bit Raspbian buster ? I will have a try immediately.
But in other raspberry pi and system versions, does it mean that the current code really cannot be compiled?
I think it may be the reason for bullseye, maybe pi3 pi3+ or pi4 have little impact.

lida2003, 2023-01-31:
> That means i need **RPI 3B+** and 32bit Raspbian **buster** ?

I mean it works for me :)

The error is caused by Makefile, you have to modify the Makefile if there is some compatible issue.

lida2003, 2023-01-31:
> * We do need to move to a newer version of IDF (5.0) . Because the support of 80211 raw radio doesn't include dumping the CRC error frame in IDF 4.3. We still need the error frames because we have [...]

This new feature is worth to implement. Can you open a new issue, maybe a PR is well appreciated.

whstudio123, 2023-01-31:
Thank you for your warm support and truxt.After i successfully compile the gs code,i will make a new issue foucus on move to IDF 5.0. 
I also think we need new system (Bullseye) and 64 bit support to make more speed , compatibility and flexibility.
The work foucus on makefile and a few lines of code need to be changed because of some ptr issues (I have done some experiment and modification.).Do you think we also need this PR?

lida2003, 2023-01-31:
Frankly speaking, currently,  gs compiles ok, but it doesn't work properly #20  with airunit (IDF 4.3 or 4.4 latest or 4.3 beta)

I have got gs compiled on RPI 3B+(minimummodification), and I also have ubuntu x86 compiled. BUT both applicaiton on PI or x86 shows black screen(I'm afraid).

Try to get the rootcause of black screen first.

whstudio123, 2023-01-31:
Here comes the result .I successfully compiled the gs code **in 32bit buster with rpi4**.
That means the code needs buster version in rpi platform.
Now I will first foucus on getting the rootcause of black screen and then have a check with the Makefile compatibility.

whstudio123, 2023-02-01:
@lida2003  I heard that you have compiled code in ubuntu linux. What is your system version? I have been working on ubuntu 18.04 64bit and undefied reference happened. I would appreciate it If you could help me to find something in makefiles.

lida2003, 2023-02-01:
Please use this patch, currently not accepted by @jeanlemotan , BUT it should work. [Fix compatibility issue in Makefie on x86_64 ubuntu 20.04](https://github.com/jeanlemotan/esp32-cam-fpv/pull/25)

whstudio123, 2023-02-01:
When you are so smart and diligent in modifying your code, I am also looking for problems in the makefile and source code. In fact, the points we noticed are all the same (nice!). I will add several links below to help later people understand the reasons for these Makefiles changes:

whstudio123, 2023-02-01:
I have reviewed the code and make some comments.