jeanlemotan/esp32-cam-fpv #27: Milestone for airunit-firmware and gs desktop program (requested.)

lida2003, 2023-02-03:
I think esp32-cam-fpv need a milestone, which should be at least
a) both airunit-firmware and gs program can be compiled.
b) both airunit-firmware (AI thinker OV2604) + RPI 3B/3B+/4 should be OK
c) both airunit-firmware (AI thinker OV2604) + laptop (ubuntu 64) should be OK
d) FPS rate ~ 30, resolution 800 x 600??? ===> which is usable for FPV
e) IDF 4.4.3 ???? IDF 5.0 ???? I think 4.4.3 might be ok for initial step.
f) others....... 

Please feel free to discuss?

whstudio123, 2023-02-04:
You are right .We need a milestone to guarantee the basic user experience.
I think that a , b and c in the list is not a problem if your Makefile modify PR is accepted.

For d,  There may be some limitations on ov2640 because of its low fps.  Go and check its datasheet .May add some codes to support OV5640.( The code is just about at some resolution micros.The ESP32-CAM officially support OV5640 )
It meet the fps of 1280*720@60 and there are thousands and hundreds of OV5640 in taobao which fit ESP32-CAM.
I have bought one.

For e. I think that it is enough for now.

whstudio123, 2023-02-04:
Don't forget that we still need some works on the ui. I'm a beginner with open gl and imgui.The gl_flush() code  could not just forever stay in there. We need to  use imgui demo as an example and rewrite some code.

lida2003, 2023-02-05:
> It meet the fps of 1280*720@60 and there are thousands and hundreds of OV5640 in taobao which fit ESP32-CAM.\nI have bought one.

Did you try?

jeanlemotan, 2023-02-05:
IDF4.4.4 should be fine now, just tested it. 5.0 will be tricky to support and TBH I don't see why. Is there anything interesting in it for this project?
I'll test tomorrow the GS black screen issue.

lida2003, 2023-02-06:
> DF4.4.4 should be fine now, just tested it.

It's OK with IDF 4.4.4, actually sdkconfig can be deleted, it's generated by menuconfig automatically.  

> the GS black screen issue.
It has something to do with glFlush, BUT video quality is BAD, please checkout the screen in latest code(9610792） no video on RPI Window #22 

> Is there anything interesting in it for this project?

Yes. a lot of camera has been added in IDF 5.0. And we also has a hardware wanna try this FPV video, which is based on IDF 5.0.

jeanlemotan, 2023-02-06:
Fixed the sdkconfig here: https://github.com/jeanlemotan/esp32-cam-fpv/commit/29fd749868ab9e18ebad0e45dd18958e9bd71e91

jeanlemotan, 2023-02-06:
Committed a bunch of air and gs fixes. Try again pls.
I'm getting good FPS and low latency again.
The memory situation in the new IDF (4.4.4) is not ideal, I had to lower some send buffers bu 10K (from 70K to 60K) to avoid the wifi driver getting stuck in receiving messages.

jeanlemotan, 2023-02-06:
Also note that the TX rate & resolution have a huge impact over FPS: if the rate is too low for the bandwidth needed, you'll get a lot of packets lost since buffers will overflow.
This is indicated by this number here:
[image]
This indicates how much of the send buffer is full and pending sending: too big (>60-70%) and you're running the risk of overflows as the wifi is not fast enough to keep up with the data flow from the camera.