jeanlemotan/esp32-cam-fpv #29: Mouse Y-Axis Offset bug

whstudio123, 2023-02-04:
In fact, in my computer, the mouse pointer has some pixel offset on the y-axis. The offset of these pixels is exactly equal to the height of the taskbar at the top of ubuntu.
 I know this should be a question of imgui. Because I have also seen similar problems in imgui. Does the program have mouse drift on your computer? Can anyone help solve this problem?

jeanlemotan, 2023-02-05:
imgui uses screen-space instead of window-space coordinates, so the fix is to subtract the window position from the mouse x and y. Let me check and I'll post a PR

jeanlemotan, 2023-02-05:
Looks like a mgui-sdl bug. Found this: https://github.com/ocornut/imgui/issues/2151

closed 2023-02-05 by jeanlemotan