jeanlemotan/esp32-cam-fpv #30: FPS and stucky problems

whstudio123, 2023-02-04:
Have anyone really could enjoy an f 
ps higher than 10? Or just some very stucky things? 

Please post your system info , device info , WiFi card type, driver version and download address.
We need to find the best combination .
 I'm suffering from super looooow fps with both of my monitor card. I have checked the raw pack in tcpdump bytes by bytes and figured out my cards really lost a lot of packcts.

I have borrowed an EM wave darkroom(a large shield box with materials whcih could absorb EM wave) from my college and do more test to.see if it is heavy traffic that influence.

jeanlemotan, 2023-02-05:
What cards do you have? Monitor mode is tricky with many cards dropping packets like crazy, or limiting the rate or just being buggy. This is why I recommend to stick to the tested wifi adapters from  EZ-wifibroadcast  or similar projects, and check any notes they might have about your cards. You might have to fix the driver, recompile the kernel or try another type of adapter...

whstudio123, 2023-02-06:
What you said is really reasonable. Not all cards have complete monitor functions. I now use the same network card model as you, and try the rtl8812au network card model in openHD. **I can now use two network cards to achieve the smoother effect you said.**
But through further analysis of tcpdump, the biggest discovery is that **the noisy electromagnetic environment** leads to complete packet loss and the existence of fcsfail frames (it seems that the transmission power of esp32 is not as strong as that of the enterprise-level APs arranged every 5m above my head).

**I have tried ESP32 with cards in EM-wave dark room.they works much better with single card**.Though i can see about 20% of packets loss in tcpdump(That maybe about the driver but not wireless traffic.)

lida2003, 2023-02-15:
I think the problem is caused by internal PCB antenna. 

I tried 8812AU cards and PCB antenna, VFPS is not good, really low about 4~5 VFPS.

Maybe esp32 should use ips external antenna.

Ncerzzk, 2023-02-18:
Hi so the disscusion above is under the resolution(800X600) ? 
if yes, I test with my **AR9271** (single) in my linux laptop and orangepi3, the vfps are all less then 10

lida2003, 2023-02-19:
@Ncerzzk  Close range test (LOS, about 30cm ???)
a) two 8812AU with kms driver + laptop wifi card, 3 wifi card total
b) 800x600(SVGA)  quality 50
c) laptop (400~500 FPS)
d) external ipx 3dbi antenna (0ohm resistor is NOT very well mounted, maybe some dbi signal loss)

ESP32 prints 40FPS
gs shows vfps  around 15 vfps 

PS: There are about 15 ssid around, very noisy environment).

Ncerzzk, 2023-02-19:
@lida2003 add the fec_n to make the link stronger could improve vfps.
I use fec_k=2, fec_n=6, and single AR9271, 640*480, vfps is range from 10 ~ 19(about 0.5m distance)
while limited by wifi rates, fec_n couldn't be set to 6, and can only get about 5~10 vpfs.

wifi env:
[image]

jeanlemotan, 2023-02-20:
What rate are you using? What % of the wifi buffer is being used? check the esp32 log, it prints there the send buffer usage %.

lida2003, 2023-02-21:
> It looks like you're getting severe packet loss. 

Yes.

>If well tuned, you should get the same VFPS as the ESP is reporting. What rate are you using?

@jeanlemotan Nice tips.

https://www.bilibili.com/video/BV1vs4y1h742/

@Ncerzzk Thanks for wifi channel scanner. I use channel 4 for test.

[image]

> What % of the wifi buffer is being used? check the esp32 log, it prints there the send buffer usage %.

I think rate/channel/pcb antenna  caused low FPS rate.

Ncerzzk, 2023-02-21:
@lida2003 @jeanlemotan 
Thank you. I finnally got the same VFPS as ESP reported (800*600,30vfps) by changing to a cleaner channel(4) in such condition:

1. default fec arguments(k=2,n=3)
2. AR9271 card（I don't know the dbi of antenna,may be 3dbi)
3. RATE_G_24M_ODFM rate
4. ESP32 AITHINKER board(with 3dbi external antenna)

so the most important factor to influenced the perf is the electromagnetic environment.
noisy electromagnetic environment -> low link quality -> lots of packets loss -> low vfps

Ncerzzk, 2023-02-21:
may be we can add a feature in esp32 to auto choose the cleanest channel at start time or run time

whstudio123, 2023-02-21:
yeah ，emwave influence a lot 。
I just get esp32 closer and a high fps

whstudio123, 2023-02-21:
2.4GHz has a very busy traffic. Some aps send beacons which may have collision with esp32.(This could be seen by using tcpdump with "fcsfail on"monitor mode) Although they obey the listen before talk rule,these aps don't receive our signal beacuse our poor tx power and dirctional ant.So they just send and both the beacon and our packet are CRC failed.

lida2003, 2023-02-22:
> I finnally got the same VFPS as ESP reported (800*600,30vfps)

All linux laptop and orangepi3 reached 30vfps? or just laptop, how about orangePi3?

Ncerzzk, 2023-02-22:
> > I finnally got the same VFPS as ESP reported (800*600,30vfps)
> 
> @Ncerzzk All linux laptop and orangepi3 reached 30vfps? or just laptop, how about orangePi3?

@lida2003 laptop and orangepi3 can all reach 30vfps, and sometimes it would drop to 10+vfps due to the noisy, anyway, it work much better than beform. I will add more info about platform above.