jeanlemotan/esp32-cam-fpv #31: IDF4.4.4 sd_enqueue_proc stack issue

lida2003, 2023-02-06:
I use the IDF4.4.4 with this default SDK config.

1024 might be not enough for sd_enqueue_proc, 1536 seems fine.
`BaseType_t res = xTaskCreatePinnedToCore(&sd_enqueue_proc, "SD Enq", 1024, nullptr, 1, &s_sd_enqueue_task, core);`

```
$ idf.py -p /dev/ttyUSB0 flash monitor
Executing action: flash
Running ninja in directory /home/daniel/Work/esp32-cam-fpv/air_firmware/build
Executing "ninja flash"...
[1/5] cd /home/daniel/Work/esp32-cam-fpv/air_firmware/build/esp-idf/esptool_py && ...tion-table.bin /home/daniel/Work/esp32-cam-fpv/air_firmware/build/air_firmware.bin
air_firmware.bin binary size 0xb9010 bytes. Smallest app partition is 0x100000 bytes. 0x46ff0 bytes (28%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/daniel/Work/esp32-cam-fpv/air_firmware/build/bootloader/esp-idf/esptool_py && /home/daniel/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/daniel/Work/esp32-cam-fpv/air_firmware/build/bootloader/bootloader.bin
Bootloader binary size 0x63b0 bytes. 0xc50 bytes (11%) free.
[2/3] cd /home/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esptool_py && /usr/bi...ome/daniel/Work/esp-sdk/esp-idf-v4.4.4/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 air_firmware.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3.2
Serial port /dev/ttyUSB0
Connecting..........
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:5a:1b:6b:d5:74
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x000c9fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 25520 bytes to 15998...
Writing at 0x00001000... (100 %)
Wrote 25520 bytes (15998 compressed) at 0x00001000 in 1.2 seconds (effective 168.2 kbit/s)...
Hash of data verified.
Compressed 757776 bytes to 474248...
Writing at 0x00010000... (3 %)
Writing at 0x0001c116... (6 %)
Writing at 0x0002760a... (10 %)
Writing at 0x0003200b... (13 %)
Writing at 0x000376a1... (17 %)
Writing at 0x0003d876... (20 %)
Writing at 0x0004343b... (24 %)
Writing at 0x0004923c... (27 %)
Writing at 0x0004ec06... (31 %)
Writing at 0x0005482c... (34 %)
Writing at 0x00059e61... (37 %)
Writing at 0x0005f118... (41 %)
Writing at 0x00064504... (44 %)
Writing at 0x000698ca... (48 %)
Writing at 0x0006ef0d... (51 %)
Writing at 0x0007483c... (55 %)
Writing at 0x0007a77a... (58 %)
Writing at 0x000801b9... (62 %)
Writing at 0x00085a47... (65 %)
Writing at 0x0008b169... (68 %)
Writing at 0x00090829... (72 %)
Writing at 0x00096c96... (75 %)
Writing at 0x0009c716... (79 %)
Writing at 0x000a4812... (82 %)
Writing at 0x000ad347... (86 %)
Writing at 0x000b3020... (89 %)
Writing at 0x000b87d3... (93 %)
Writing at 0x000be1e2... (96 %)
Writing at 0x000c3cf7... (100 %)
Wrote 757776 bytes (474248 compressed) at 0x00010000 in 11.7 seconds (effective 517.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
[... 397 lines cut ...]
I (1442) camera: Allocating DMA buffer #1, size=3200
I (1442) camera: Allocating DMA buffer #2, size=3200
I (1452) camera: Allocating DMA buffer #3, size=3200
I (1452) camera: Allocating DMA buffer #4, size=3200
I (1462) camera: Allocating DMA buffer #5, size=3200
I (1462) camera: Allocating DMA buffer #6, size=3200
I (1472) camera: Allocating DMA buffer #7, size=3200
MEMORY Before Loop:
Heap summary for capabilities 0x00000004:
  At 0x3ffae6e0 len 6432 free 4 allocated 5912 min_free 4
    largest_free_block 0 alloc_blocks 34 free_blocks 0 total_blocks 34
  At 0x3ffb8a38 len 161224 free 4 allocated 159092 min_free 4
    largest_free_block 0 alloc_blocks 297 free_blocks 0 total_blocks 297
  At 0x3ffe0440 len 15072 free 4 allocated 14444 min_free 4
    largest_free_block 0 alloc_blocks 52 free_blocks 0 total_blocks 52
  At 0x3ffe4350 len 113840 free 4100 allocated 108468 min_free 2464
    largest_free_block 1600 alloc_blocks 100 free_blocks 3 total_blocks 103
  Totals:
    free 4112 allocated 287916 min_free 2476 largest_free_block 1600
S: 87084, R: 0, E: 0, D: 0, % : 97 | FPS: 1, D: 133577 | D: 0, E: 0
S: 1359396, R: 0, E: 0, D: 0, % : 97 | FPS: 39, D: 3066468 | D: 0, E: 0
S: 1526184, R: 0, E: 0, D: 0, % : 95 | FPS: 40, D: 1909868 | D: 0, E: 0
S: 1490760, R: 0, E: 0, D: 0, % : 97 | FPS: 40, D: 1743754 | D: 0, E: 0

```

jeanlemotan, 2023-02-06:
Fixed in https://github.com/jeanlemotan/esp32-cam-fpv/commit/29fd749868ab9e18ebad0e45dd18958e9bd71e91

closed 2023-02-06 by jeanlemotan

lida2003, 2023-02-07:
Is this airfirmware should be upgrde with gs together?

Since my old airunit firmware works with gs(laptop version), and it give black screen when use this fixed version.

```
commit 3dfd013147004bfb3277d8999f9d58e77673b0c6 (HEAD -> jeanlemotan-main, origin/jeanlemotan-main, origin/HEAD)
Author: jeanlemotan <10252034+jeanlemotan@users.noreply.github.com>
Date:   Mon Feb 6 10:35:58 2023 +0100

    GS Fixes (contributed)
```