jeanlemotan/esp32-cam-fpv #36: Laptop build

Yury-MonZon, 2023-03-07:
Hi there! Pop OS 6.1.11-76060111-generic #202302091138~1675975749~22.04~f771a7f SMP PREEMPT_DYNAMIC Thu F x86_64 x86_64 x86_64 GNU/Linux here. HW: TL-WN722N v2 card with [this fix](https://github.com/gglluukk/rtl8188eus). Airdump-ng works fine - tested it.

`
$ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 286: Drivers: 4
(I) src/PI_HAL.cpp: 289: Driver 0: x11
(I) src/PI_HAL.cpp: 289: Driver 1: wayland
(I) src/PI_HAL.cpp: 289: Driver 2: KMSDRM
(I) src/PI_HAL.cpp: 289: Driver 3: dummy
(I) src/PI_HAL.cpp: 298: Mode -1: 455844177x671522212
(E) src/PI_HAL.cpp: 317: Cannot create window: No available video device
(E) src/PI_HAL.cpp: 604: Cannot initialize display
Segmentation fault
` 
or

`
 sudo -E ./gs
(I) src/PI_HAL.cpp: 286: Drivers: 4
(I) src/PI_HAL.cpp: 289: Driver 0: x11
(I) src/PI_HAL.cpp: 289: Driver 1: wayland
(I) src/PI_HAL.cpp: 289: Driver 2: KMSDRM
(I) src/PI_HAL.cpp: 289: Driver 3: dummy
(I) src/PI_HAL.cpp: 298: Mode 0: 1366x768
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile vertex shader!
0:1(10): error: GLSL 1.30 is not supported. Supported versions are: 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile fragment shader!
0:1(10): error: GLSL 1.30 is not supported. Supported versions are: 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
(E) src/imgui_impl_opengl3.cpp: 382: GL error 1282 in glShaderSource(shaderData.VertHandle, 2, vertex_shader_with_version, NULL) file src/imgui_impl_opengl3.cpp line 382
ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile vertex shader!
0:1(10): error: GLSL 1.30 is not supported. Supported versions are: 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile fragment shader!
0:1(10): error: GLSL 1.30 is not supported. Supported versions are: 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
(E) src/imgui_impl_opengl3.cpp: 477: GL error 1282 in glGenBuffers(1, &g_VboHandle) file src/imgui_impl_opengl3.cpp line 477
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlxac15a2eb4253 in monitor mode
(E) src/Comms.cpp: 623: Error in pcap_activate: That device doesn't support monitor mode
`
What's my problem there? 
Thanks

jeanlemotan, 2023-03-07:
The problem seems to be this:
`0:1(10): error: GLSL 1.30 is not supported. Supported versions are: 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES`
Can you comment this code and try?
```cpp
    if (glsl_version == NULL)
        glsl_version = "#version 130";
    IM_ASSERT((int)strlen(glsl_version) + 2 < IM_ARRAYSIZE(g_GlslVersionString));
    strcpy(g_GlslVersionString, glsl_version);
    strcat(g_GlslVersionString, "\n");
```

It's in the imgui_impl_opengl3.cpp file, line 107.

Yury-MonZon, 2023-03-07:
Commented this block out.
And I get this:

`$ sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 286: Drivers: 4
(I) src/PI_HAL.cpp: 289: Driver 0: x11
(I) src/PI_HAL.cpp: 289: Driver 1: wayland
(I) src/PI_HAL.cpp: 289: Driver 2: KMSDRM
(I) src/PI_HAL.cpp: 289: Driver 3: dummy
(I) src/PI_HAL.cpp: 298: Mode -1: 947570897x1420212923
(E) src/PI_HAL.cpp: 317: Cannot create window: No available video device
(E) src/PI_HAL.cpp: 604: Cannot initialize display
Segmentation fault

$ sudo -E ./gs
(I) src/PI_HAL.cpp: 286: Drivers: 4
(I) src/PI_HAL.cpp: 289: Driver 0: x11
(I) src/PI_HAL.cpp: 289: Driver 1: wayland
(I) src/PI_HAL.cpp: 289: Driver 2: KMSDRM
(I) src/PI_HAL.cpp: 289: Driver 3: dummy
(I) src/PI_HAL.cpp: 298: Mode 0: 1366x768
(I) src/Comms.cpp: 718: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 581: Opening interface wlxac15a2eb4253 in monitor mode
(E) src/Comms.cpp: 623: Error in pcap_activate: That device doesn't support monitor mode
`

Ncerzzk, 2023-03-08:
can you test with `sudo tcpdump --monitor-mode -i wlxac15a2eb4253`
I'm sorry that I know little about ` Airdump-ng` 
generally in my test case, if the tcpdump operating netcard works fine, gs can work too.

Yury-MonZon, 2023-03-08:
Sadly, it doesn't work this way:
`tcpdump: wlxac15a2eb4253: That device doesn't support monitor mode` I think I can try to fix it later on.

But the main problem here is that I'm not passed the video issue yet. Would really appreciate your help here:
`sudo -E DISPLAY=:0 ./gs
(I) src/PI_HAL.cpp: 286: Drivers: 4
(I) src/PI_HAL.cpp: 289: Driver 0: x11
(I) src/PI_HAL.cpp: 289: Driver 1: wayland
(I) src/PI_HAL.cpp: 289: Driver 2: KMSDRM
(I) src/PI_HAL.cpp: 289: Driver 3: dummy
(I) src/PI_HAL.cpp: 298: Mode -1: 947570897x1420212923
(E) src/PI_HAL.cpp: 317: Cannot create window: No available video device
(E) src/PI_HAL.cpp: 604: Cannot initialize display
Segmentation fault`

Thanks a lot!

Ncerzzk, 2023-03-08:
please exec `printenv` to get your DISPLAY info first

Yury-MonZon, 2023-03-08:
Just to be sure: I should see beacons and acknowledgements, right?

Ncerzzk, 2023-03-09:
> 
why remove `--monitor-mode` option?

Yury-MonZon, 2023-03-09:
Because I have to put my card into the monitor mode via script. Can you provide some info on what tcpdump prints out on a properly working setup?

Ncerzzk, 2023-03-09:
`sudo tcpdump --monitor-mode -i wlxac15a2eb4253` just a command to test whether your netcard could be operated well.
if it runs ok, you can directly use `sudo ./gs`.

Ncerzzk, 2023-03-09:
gs didn't contain tcpdump.  tcpdump for me is just used for test netcard.

try to remove `if (pcap_set_rfmon(pcap.pcap, 1) < 0) `
https://github.com/jeanlemotan/esp32-cam-fpv/blob/adc1e5c040bada4780de0db60716af411d326435/gs/src/Comms.cpp#LL599C9-L599C23

Ncerzzk, 2023-03-10:
you can package a static jpeg into c header, then send it by call `camera_data_available` manually