jeanlemotan/esp32-cam-fpv #42: Error in pcap_activate: That device doesn't support monitor mode

krzysztofkuczek, 2024-05-10:
Hi,
I trying to make gs works on RPI4 running openhd kernel 6.1.29-v8+ and openhd rtl drivers for my TP-Link RTL8812AU Archer T4U 802.11ac wifi adapter.
Unfortunelty it does not work even when the wifi seems to be in monitor mode
```
iw wlan1 info
Interface wlan1
        ifindex 6
        wdev 0x300000001
        addr c4:e9:84:09:07:35
        type monitor
        wiphy 3
        channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
        txpower 18.00 dBm
```
The error is " Error in pcap_activate: That device doesn't support monitor mode" and tcpdump also claims the same 

```
 sudo tcpdump --monitor-mode -i wlan1
tcpdump: wlan1: That device doesn't support monitor mode
```
Any idea what i do wrong,
thank you in advance
Kris

RomanLut, 2024-05-12:
```sudo apt-get install aircrack-ng```
```sudo airmon-ng start wlan1```

Then use ```wlan1mon```.

krzysztofkuczek, 2024-05-13:
Hi,
DO you mean that another wlan1mon adapter should be available?
It's not, and wlan1 still does not work :(

 sudo airmon-ng start wlan1

PHY     Interface       Driver          Chipset

phy0    wlan0           brcmfmac        Broadcom 43430
phy1    wlan1           rtl88xxau_ohd   TP-Link RTL8812AU Archer T4U 802.11ac
                (mac80211 monitor mode already enabled for [phy1]wlan1 on [phy1]10)

kku@rpi4:~/src/esp32-cam-fpv/gs $ iw wlan1 info
Interface wlan1
        ifindex 4
        wdev 0x100000001
        addr c4:e9:84:09:07:35
        type monitor
        wiphy 1
        channel 10 (2457 MHz), width: 20 MHz, center1: 2457 MHz
        txpower 18.00 dBm
kku@rpi4:~/src/esp32-cam-fpv/gs $ sudo tcpdump --monitor-mode -i wlan1
tcpdump: wlan1: That device doesn't support monitor mode
kku@rpi4:~/src/esp32-cam-fpv/gs $ sudo -E DISPLAY=:0 ./gs
width:1280 height:720
(I) src/Comms.cpp: 723: Radiocap header size: 11, IEEE header size: 24
(I) src/Comms.cpp: 583: Opening interface wlan1 in monitor mode
(E) src/Comms.cpp: 628: Error in pcap_activate: That device doesn't support monitor mode

RomanLut, 2024-05-13:
```sudo airmon-ng start wlan1``` should create another interface ```wlan1mon``` with monitoring support.

But it seems that you are using some special drivers.

1) Just try  to remove that check from comms.cpp

2) if it does not help, I think you have to build GS image with another drivers, you can follow this steps:
https://github.com/RomanLut/esp32-cam-fpv/blob/master/doc/building_gs_image.md

krzysztofkuczek, 2024-05-17:
Thank you, now it works, but the esp32s3sense is f... hot. Even with quite big heat sink it's hot. It draws 338mAm at 5V.
Any idea?
Kris

RomanLut, 2024-05-17:
I am powering it with 4.5V on a plane, it hot but it work so far :) You can try lowering voltage up to 3.5V on VIN pin:
https://files.seeedstudio.com/wiki/SeeedStudio-XIAO-ESP32S3/res/XIAO_ESP32S3_SCH_v1.1.pdf

krzysztofkuczek, 2024-05-17:
Already tried, the below 4v, the camera module seems to have some problems, showing green horizontal line, and the current is risings, U * I seems to be constant ;) So it seems to be not related to LDO voltage drop...
Thank you.
Kris

krzysztofkuczek, 2024-05-24:
One More question,
I was trying to move from rpi4 to rpi 0 2w, compiled everything from scratch, but without success. The wlan1 is opened correctly, but no single frame received. Any hints?

RomanLut, 2024-05-24:
1) I assume you compile both GS and esp32cam firmware from the same branch. 

2) Check serial output on air unit, you should see something like this:
```
WLAN S: 738000, R: 210, E: 0, F: 0, D: 0, % : 18...37 || FPS: 29, D: 331624 || SD D: 327680, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 42, Q: 23 s: 12434
WLAN S: 792612, R: 210, E: 0, F: 0, D: 0, % : 18...39 || FPS: 30, D: 362635 || SD D: 360448, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 41, Q: 24 s: 13090
WLAN S: 692244, R: 210, E: 0, F: 0, D: 0, % : 18...37 || FPS: 29, D: 314430 || SD D: 315392, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 33, Q: 29 s: 12288
WLAN S: 712908, R: 210, E: 0, F: 0, D: 0, % : 2...19 || FPS: 29, D: 326076 || SD D: 323584, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 41, Q: 24 s: 11330
WLAN S: 777852, R: 210, E: 0, F: 0, D: 0, % : 9...30 || FPS: 29, D: 355329 || SD D: 352256, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 48, Q: 20 s: 12321
WLAN S: 746856, R: 140, E: 0, F: 0, D: 0, % : 53...65 || FPS: 29, D: 348752 || SD D: 348160, E: 0 || TLM IN: 0 OUT: 0 || SK1: 100 SK2: 100, SK3: 1, Q: 60 s: 13991
```

FPS:29 means that camera actually works.

2) Bring debug interface with middle mouse button or d key:

[image]

FPS: 27/2 in the tille means that 27 frames was decoded and 2 was lost due to packet loss.

Do you see any numbers in the FPS: line?
What is RSSI value ?

wookis1113, 2024-07-20:
i  cant enter  debug interface with middle mouse button or d key
there are too many noise in my video due to Interference by next door 
Is there anything I can do?